apuder/TRS-IO #15: dram0_0_seg size DRAM segment data does not fit.

johnwiggins, 2019-12-01:
Sorry, yet another problem loading:

LD esp/build/trs-io.elf
/home/wiggins/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/5.2.0/../../../../xtensa-esp32-elf/bin/ld: /home/wiggins/TRS-IO/src/esp/build/trs-io.elf section `.dram0.bss' will not fit in region `dram0_0_seg'
/home/wiggins/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/5.2.0/../../../../xtensa-esp32-elf/bin/ld: DRAM segment data does not fit.
/home/wiggins/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/5.2.0/../../../../xtensa-esp32-elf/bin/ld: region `dram0_0_seg' overflowed by 22872 bytes

apuder, 2019-12-01:
Now this is weird. Can you please answer a few quick questions?
- Which version of the Espressif SDK did you use?
- Did you make any modifications to sdkconfig?
- Did you make any modifications to the code of TRS-IO?

johnwiggins, 2019-12-01:
I used esp-idf 3.2.2 because the latest Git version (4?) wouldn't build the TRS-IO sources at all. The  extensa compilers are from xtensa-esp32-elf-linux64-1.22.0-80-g6c4433a-5.2.0.tar.gz.

The build was done on a fresh clone of TRS-IO with defaults selected during config. I didnt change anything in the current build tree.

apuder, 2019-12-01:
Which ESP32 variant do you use? As a possible fix, can you change the declarations in rsclient.c to "const"? (the file you were asking about in your other issue). You might have to make a change somewhere else to match the "const" declaration but the compiler will tell you. By declaring the huge array as const will prevent it from being loaded into RAM and might fix your problem. But I'm still a little unsure why I'm not seeing this problem.

johnwiggins, 2019-12-01:
I'm using a Hiletgo ESP32-WROOM-32. Changing the variables in rsclient.c to const worked without having to change anything anywhere else. eg:

 esp]$ make                                                                                    
Toolchain path: /home/wiggins/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc                                        
Toolchain version: crosstool-ng-1.22.0-80-g6c4433a                                                             
Compiler version: 5.2.0                                                                                        
Python requirements from /home/wiggins/esp/esp-idf/requirements.txt are satisfied.                             
CC build/retrostore/rsclient.o                                                                                 
AR build/retrostore/libretrostore.a                                                                            
LD build/trs-io.elf                                                                                            
esptool.py v2.6                                                                                                
To flash all build output, run 'make flash' or:                                                                
python /home/wiggins/esp/esp-idf/components/esptool_py/esptool/esptool.py --chip esp32 --port /dev/ttyS4 --baud
 115200 --before default_reset --after hard_reset write_flash -u --flash_mode dio --flash_freq 40m --flash_size
 detect 0xd000 /home/wiggins/TRS-IO/src/esp/build/ota_data_initial.bin 0x1000 /home/wiggins/TRS-IO/src/esp/buil
d/bootloader/bootloader.bin 0x10000 /home/wiggins/TRS-IO/src/esp/build/trs-io.bin 0x8000 /home/wiggins/TRS-IO/s
rc/esp/build/partitions_two_ota.bin

apuder, 2019-12-01:
Better do a "make clean" to be sure. You should also have to change it here:
https://github.com/apuder/TRS-IO/blob/master/src/esp/components/retrostore/retrostore.cpp#L18

johnwiggins, 2019-12-01:
Make clean worked too, and didn't throw an error from retrostore.cpp.

Was able to load the code from the pre-make clean build and the ESP32 booted, went into config mode and once configured, happily joined my wifi network. Still have to test the rest of the card on a M3, but so far, so good! Thank you and I owe you a beer!

apuder, 2019-12-02:
I still suggest to change the declaration in retrostore.cpp to const as well. Please keep me posted. I'm especially interested in you can get the FreHD part to work. Another user had some issues with that.

johnwiggins, 2019-12-02:
okay changing retrostore.cpp to const breaks things.

CXX build/retrostore/retrostore.o                                                                              
/home/wiggins/TRS-IO/src/esp/components/retrostore/retrostore.cpp: In member function 'void RetroStoreModule::s
endCMD()':                                                                                                     
/home/wiggins/TRS-IO/src/esp/components/retrostore/retrostore.cpp:53:47: error: invalid conversion from 'const 
void*' to 'void*' [-fpermissive]                                                                               
       addBlob16(rsclient_cmd, rsclient_cmd_len);                                                              
                                               ^                                                               |
In file included from /home/wiggins/TRS-IO/src/esp/components/retrostore/retrostore.cpp:2:0:                   |
/home/wiggins/TRS-IO/src/esp/components/trs-io/include/trs-io.h:192:24: note:   initializing argument 1 of 'sta
tic void TrsIO::addBlob16(void*, uint16_t)'                                                                    |
     inline static void addBlob16(void *blob, uint16_t len) {                                                  |
                        ^                                                                                      |
/home/wiggins/TRS-IO/src/esp/components/retrostore/retrostore.cpp:61:49: error: invalid conversion from 'const 
void*' to 'void*' [-fpermissive]                                                                               |
         addBlob16(rsclient_cmd, rsclient_cmd_len);                                                            |
                                                 ^                                                             
In file included from /home/wiggins/TRS-IO/src/esp/components/retrostore/retrostore.cpp:2:0:                   |
/home/wiggins/TRS-IO/src/esp/components/trs-io/include/trs-io.h:192:24: note:   initializing argument 1 of 'sta|
tic void TrsIO::addBlob16(void*, uint16_t)'                                                                    |
     inline static void addBlob16(void *blob, uint16_t len) {
                        ^                                                                                      |
make[1]: *** [retrostore.o] Error 1
make: *** [component-retrostore-build] Error 2

apuder, 2019-12-02:
Well, didn't consider that. Then just declare the array as const in rsclient.c since that seems to work for you. I'll work on a more general fix. Please let me know if you have any issues with FreHD. Thanks!

closed 2019-12-02 by apuder