apuder/TRS-IO #18: esp32 app crashes

iceblu3710, 2020-01-05:
I got esp-idf v3.2.2 installed and can load the hello world example successfully. 

I ran make in the src folder and was asked a dozen questions I know nothing about. I do know if you select release instead of debug you get compile errors. I stayed with debug and it compiled. I then went to src/esp and did a make menuconfig, set the TTY and then make flash monitor and I get a esp32 program crashing.

````I (5318) cpu_start: Pro cpu up.
I (5318) cpu_start: Starting app cpu, entry point is 0x40081010
0x40081010: call_start_cpu1 at /home/uminded/esp/esp-idf/components/esp32/cpu_start.c:246

I (5304) cpu_start: App cpu up.
D (5321) memory_layout: Checking 7 reserved memory ranges:
D (5327) memory_layout: Reserved memory range 0x3ffae000 - 0x3ffae6e0
D (5333) memory_layout: Reserved memory range 0x3ffb0000 - 0x3ffd8450
D (5340) memory_layout: Reserved memory range 0x3ffe0000 - 0x3ffe0440
D (5346) memory_layout: Reserved memory range 0x3ffe3f20 - 0x3ffe4350
D (5353) memory_layout: Reserved memory range 0x40070000 - 0x40078000
D (5359) memory_layout: Reserved memory range 0x40078000 - 0x40080000
0x40080000: _WindowOverflow4 at /home/uminded/esp/esp-idf/components/freertos/xtensa_vectors.S:1779

D (5366) memory_layout: Reserved memory range 0x40080000 - 0x40090094
0x40080000: _WindowOverflow4 at /home/uminded/esp/esp-idf/components/freertos/xtensa_vectors.S:1779

D (5372) memory_layout: Building list of available memory regions:
D (5379) memory_layout: Available memory region 0x3ffae6e0 - 0x3ffb0000
D (5385) memory_layout: Available memory region 0x3ffd8450 - 0x3ffda000
D (5392) memory_layout: Available memory region 0x3ffda000 - 0x3ffdc000
D (5399) memory_layout: Available memory region 0x3ffdc000 - 0x3ffde000
D (5405) memory_layout: Available memory region 0x3ffde000 - 0x3ffe0000
D (5412) memory_layout: Available memory region 0x3ffe0440 - 0x3ffe3f20
D (5419) memory_layout: Available memory region 0x3ffe4350 - 0x3ffe8000
D (5425) memory_layout: Available memory region 0x3ffe8000 - 0x3fff0000
D (5432) memory_layout: Available memory region 0x3fff0000 - 0x3fff8000
D (5439) memory_layout: Available memory region 0x3fff8000 - 0x3fffc000
D (5445) memory_layout: Available memory region 0x3fffc000 - 0x40000000
D (5452) memory_layout: Available memory region 0x40090094 - 0x40092000
D (5459) memory_layout: Available memory region 0x40092000 - 0x40094000
D (5465) memory_layout: Available memory region 0x40094000 - 0x40096000
D (5472) memory_layout: Available memory region 0x40096000 - 0x40098000
D (5479) memory_layout: Available memory region 0x40098000 - 0x4009a000
D (5485) memory_layout: Available memory region 0x4009a000 - 0x4009c000
D (5492) memory_layout: Available memory region 0x4009c000 - 0x4009e000
D (5499) memory_layout: Available memory region 0x4009e000 - 0x400a0000
I (5506) heap_init: Initializing. RAM available for dynamic allocation:
D (5513) heap_init: New heap initialised at 0x3ffae6e0
I (5518) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM
D (5524) heap_init: New heap initialised at 0x3ffd8450
I (5529) heap_init: At 3FFD8450 len 00007BB0 (30 KiB): DRAM
I (5536) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (5542) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
D (5549) heap_init: New heap initialised at 0x40090094
I (5554) heap_init: At 40090094 len 0000FF6C (63 KiB): IRAM
I (5560) cpu_start: Pro cpu start user code
D (5572) clk: RTC_SLOW_CLK calibration value: 3409152
D (213) intr_alloc: Connected src 46 to int 2 (cpu 0)
D (214) intr_alloc: Connected src 57 to int 3 (cpu 0)
D (215) intr_alloc: Connected src 24 to int 9 (cpu 0)
I (219) cpu_start: Starting scheduler on PRO CPU.
D (0) intr_alloc: Connected src 25 to int 2 (cpu 1)
I (4) cpu_start: Starting scheduler on APP CPU.
D (243) heap_init: New heap initialised at 0x3ffe0440
D (249) heap_init: New heap initialised at 0x3ffe4350
D (254) intr_alloc: Connected src 16 to int 12 (cpu 0)
I (259) gpio: GPIO[4]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 1| Intr:0 
I (268) gpio: GPIO[5]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 1| Intr:0 
I (277) gpio: GPIO[21]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 1| Intr:0 
I (287) gpio: GPIO[22]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:3 
D (296) intr_alloc: Connected src 22 to int 13 (cpu 0)
D (302) nvs: nvs_flash_init_custom partition=nvs start=9 count=4
D (339) nvs: nvs_open_from_partition retrostore 1
D (341) nvs: nvs_open_from_partition misc 1
D (341) nvs: nvs_get_str_or_blob log
[... 120 lines cut ...]

A2      : 0x3ff4400c  A3      : 0x00000000  A4      : 0x00060023  A5      : 0x3ffd8f74  
A6      : 0x00000000  A7      : 0x00000000  A8      : 0x3ff44000  A9      : 0x08000000  
A10     : 0x00000000  A11     : 0x00060023  A12     : 0x00060023  A13     : 0x00000001  
A14     : 0x00060021  A15     : 0x00060123  SAR     : 0x00000000  EXCCAUSE: 0x00000005  
EXCVADDR: 0x00000000  LBEG    : 0x00000000  LEND    : 0x00000000  LCOUNT  : 0x00000000  

Backtrace: 0x400d5f5a:0x3ffea1b0 0x400871cd:0x3ffea1d0
0x400d5f5a: io_task(void*) at /home/uminded/Programming/esp32/TRS-IO/src/esp/main/io.cpp:156

0x400871cd: vPortTaskWrapper at /home/uminded/esp/esp-idf/components/freertos/port.c:403

Rebooting...
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:5400
load:0x40078000,len:10868
load:0x40080400,len:6412
entry 0x40080758
````

iceblu3710, 2020-01-05:
Repeating the process but with a new git clone of TRS-IO I get a little further.

I can connect to the AP and I get the IP 192.168.4.2 but TRS-IO.local or 192.168.4.1 say connection refused.

````
MONITOR
--- idf_monitor on /dev/ttyUSB0 115200 ---
--- Quit: Ctrl+] | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H ---
ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:5400
load:0x40078000,len:10868
load:0x40080400,len:6412
entry 0x40080758
I (318) cpu_start: Pro cpu up.
I (318) cpu_start: Starting app cpu, entry point is 0x40081014
0x40081014: call_start_cpu1 at /home/uminded/esp/esp-idf/components/esp32/cpu_start.c:246

I (0) cpu_start: App cpu up.
D (320) memory_layout: Checking 7 reserved memory ranges:
D (326) memory_layout: Reserved memory range 0x3ffae000 - 0x3ffae6e0
D (332) memory_layout: Reserved memory range 0x3ffb0000 - 0x3ffd8c50
D (338) memory_layout: Reserved memory range 0x3ffe0000 - 0x3ffe0440
D (345) memory_layout: Reserved memory range 0x3ffe3f20 - 0x3ffe4350
D (351) memory_layout: Reserved memory range 0x40070000 - 0x40078000
D (358) memory_layout: Reserved memory range 0x40078000 - 0x40080000
0x40080000: _WindowOverflow4 at /home/uminded/esp/esp-idf/components/freertos/xtensa_vectors.S:1779

D (364) memory_layout: Reserved memory range 0x40080000 - 0x40090098
0x40080000: _WindowOverflow4 at /home/uminded/esp/esp-idf/components/freertos/xtensa_vectors.S:1779

D (371) memory_layout: Building list of available memory regions:
D (377) memory_layout: Available memory region 0x3ffae6e0 - 0x3ffb0000
D (383) memory_layout: Available memory region 0x3ffd8c50 - 0x3ffda000
D (390) memory_layout: Available memory region 0x3ffda000 - 0x3ffdc000
D (397) memory_layout: Available memory region 0x3ffdc000 - 0x3ffde000
D (403) memory_layout: Available memory region 0x3ffde000 - 0x3ffe0000
D (410) memory_layout: Available memory region 0x3ffe0440 - 0x3ffe3f20
D (416) memory_layout: Available memory region 0x3ffe4350 - 0x3ffe8000
D (423) memory_layout: Available memory region 0x3ffe8000 - 0x3fff0000
D (430) memory_layout: Available memory region 0x3fff0000 - 0x3fff8000
D (436) memory_layout: Available memory region 0x3fff8000 - 0x3fffc000
D (443) memory_layout: Available memory region 0x3fffc000 - 0x40000000
D (449) memory_layout: Available memory region 0x40090098 - 0x40092000
D (456) memory_layout: Available memory region 0x40092000 - 0x40094000
D (463) memory_layout: Available memory region 0x40094000 - 0x40096000
D (469) memory_layout: Available memory region 0x40096000 - 0x40098000
D (476) memory_layout: Available memory region 0x40098000 - 0x4009a000
D (482) memory_layout: Available memory region 0x4009a000 - 0x4009c000
D (489) memory_layout: Available memory region 0x4009c000 - 0x4009e000
D (496) memory_layout: Available memory region 0x4009e000 - 0x400a0000
I (502) heap_init: Initializing. RAM available for dynamic allocation:
D (509) heap_init: New heap initialised at 0x3ffae6e0
I (514) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM
D (521) heap_init: New heap initialised at 0x3ffd8c50
I (526) heap_init: At 3FFD8C50 len 000073B0 (28 KiB): DRAM
I (532) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (538) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
D (545) heap_init: New heap initialised at 0x40090098
I (550) heap_init: At 40090098 len 0000FF68 (63 KiB): IRAM
I (556) cpu_start: Pro cpu start user code
D (568) clk: RTC_SLOW_CLK calibration value: 3421069
D (130) intr_alloc: Connected src 46 to int 2 (cpu 0)
D (131) intr_alloc: Connected src 57 to int 3 (cpu 0)
D (132) intr_alloc: Connected src 24 to int 9 (cpu 0)
[... 72 lines cut ...]
E (15176) task_wdt: Tasks currently running:
E (15176) task_wdt: CPU 0: main
E (15176) task_wdt: CPU 1: io
E (20176) task_wdt: Task watchdog got triggered. The following tasks did not reset the watchdog in time:
E (20176) task_wdt:  - IDLE0 (CPU 0)
E (20176) task_wdt: Tasks currently running:
E (20176) task_wdt: CPU 0: main
E (20176) task_wdt: CPU 1: io
E (25176) task_wdt: Task watchdog got triggered. The following tasks did not reset the watchdog in time:
E (25176) task_wdt:  - IDLE0 (CPU 0)
E (25176) task_wdt: Tasks currently running:
E (25176) task_wdt: CPU 0: main
E (25176) task_wdt: CPU 1: io
E (30176) task_wdt: Task watchdog got triggered. The following tasks did not reset the watchdog in time:
E (30176) task_wdt:  - IDLE0 (CPU 0)
E (30176) task_wdt: Tasks currently running:
E (30176) task_wdt: CPU 0: main
E (30176) task_wdt: CPU 1: io
E (35176) task_wdt: Task watchdog got triggered. The following tasks did not reset the watchdog in time:
E (35176) task_wdt:  - IDLE0 (CPU 0)
E (35176) task_wdt: Tasks currently running:
E (35176) task_wdt: CPU 0: main
E (35176) task_wdt: CPU 1: io
E (40176) task_wdt: Task watchdog got triggered. The following tasks did not reset the watchdog in time:
````

apuder, 2020-01-05:
It seems the HTTP task is not running which is why you cannot connect. But you shouldn't get those watchdog timers either. What is the precise ESP32 model you are using?

iceblu3710, 2020-01-05:
The only one easily available was Digikey 1965-1003-ND. ESP32-WROOM-32D

iceblu3710, 2020-01-17:
Any ideas? The 32 and 32D are the same chip just a different die revision and layout under the can.

apuder, 2020-01-17:
This is the ESP32 version I am using:
https://www.amazon.com/gp/product/B0718T232Z/ref=ppx_yo_dt_b_search_asin_title?ie=UTF8&psc=1

I know that @jaquinn also had some issues with other ESP32 flavors. If possible, could you try the version I am using?

jaquinn, 2020-02-19:
There is a pending pull request which updates the build documentation to confirm the supported ESP32 modules and the orientation on the board.  

Interestingly, I had similar issues with the Task Watchdog and it was due to omitting to solder of pin 1 on both the 74LVC245.  Have a close look at your board to be sure the solder joints are all OK.

vodkaflip, 2020-12-15:
Bumping this older issue: on the M1 version, I get the 48k expansion (holding down break). So that part works... The ESP32 however connects to the wifi (SSID and password hardcoded), starts Mongoose but then along the way, wifi disconnects (reason 8, assoc leave) and I get a SW_CPU_RESET) and the whole cycle starts again.

`I (2347) TRS-IO: Starting Mongoose
D (2347) nvs: nvs_get_str_or_blob tz
D (2349) nvs: nvs_get tz 4
D (2356) nvs: nvs_get_str_or_blob ssid
D (2357) nvs: nvs_get_str_or_blob ssid
D (2359) nvs: nvs_get_str_or_blob passwd
D (2373) nvs: nvs_get_str_or_blob passwd
D (2374) nvs: nvs_get_str_or_blob tz
D (2374) nvs: nvs_get tz 4
D (2375) nvs: nvs_get_str_or_blob smb_url
D (2378) nvs: nvs_get_str_or_blob smb_url
D (2382) nvs: nvs_get_str_or_blob smb_user
D (2386) nvs: nvs_get_str_or_blob smb_user
D (2390) nvs: nvs_get_str_or_blob smb_passwd
D (2394) nvs: nvs_get_str_or_blob smb_passwd
D (2557) phy_init: wifi mac time delta: 6616
D (2779) phy_init: wifi mac time delta: 31883
D (2842) phy_init: wifi mac time delta: 4254
D (2945) phy_init: wifi mac time delta: 30471
D (3047) phy_init: wifi mac time delta: 86897
D (3150) phy_init: wifi mac time delta: 88863
D (3152) nvs: nvs_erase_all

I (3160) wifi:state: run -> init (0)
I (3161) wifi:pm stop, total sleep time: 602049 us / 1559244 us

I (3162) wifi:new:<2,0>, old:<2,0>, ap:<255,255>, sta:<2,0>, prof:1
D (3165) event: SYSTEM_EVENT_STA_DISCONNECTED, ssid:SECTOR7g, ssid_len:8, bssid:04:d9:f5:b3:d9:e0, reason:8,wifi reason: assoc leave
D (3178) tcpip_adapter: if0 start ip lost tmr: enter
D (3182) tcpip_adapter: if0 start ip lost tmr: interval=120
D (3188) event: SYSTEM_EVENT_STA_STOP
I (3202) wifi:flush txq
I (3202) wifi:stop sw txq
D (3203) phy_init: wifi mac time delta: 64
I (3204) wifi:lmac stop hw txq
D (3205) RTC_MODULE: Wi-Fi returns adc2 lock.
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)`

I had some trouble compiling the source until I downgraded ESP_IDF to version 3.3.4. I am using SDCC 4.0 - could that be the issue (I have to change one line as sdcclib no longer exists)

apuder, 2021-12-25:
I'm closing this issue. TRS-IO now requires ESP-IDF 4.2.2. If you have issues with the latest version, please re-open this issue.

closed 2021-12-25 by apuder