apuder/TRS-IO #23: Autoboot disks don't work, only autoboot eproms.

ngtwolf, 2020-03-29:
This is less of an issue and more of a feature request.  Unlike the FreHD, the TRS-IO doesn't auto booting from the autoboot disks like a FreHD does, it only appears to work with an autoboot eprom.  Would be nice to allow it to work with autoboot disks as well, for those who don't want to (or can't easily) update replace the rom in their trs-80.

apuder, 2020-04-17:
I am not sure how the autoboot disk works. I would assume that it uses the same bootstrap mechanism as the ROM patch. That does not seem to be the case. Do you have the source to the autoboot disk? Unfortunately I do not have floppy drives so I cannot test it myself but I'd like to check the source.

kwadsten, 2020-04-17:
Not sure if this is relevant or not, but this BASIC program is supposed to be the equivalent of the FreHD autoboot ROM...

FreHD BASIC Loader

10 POKE 16912,56:OUT 236,56
20 OUT 197,3:REM 1=M1, 3=M3, 4=M4/M4A, 5=M4P
30 IF INP(196)<>254 STOP
40 FOR I=0 TO 255:POKE 20480+I,INP(196):NEXT

SYSTEM
/20480

apuder, 2020-04-17:
Actually, you can add those lines here:
`50 POKE16526,0
60 POKE16527,80
70 PRINT USR(0)`
This way you do not have to use the SYSTEM command. You should be able to turn this into an autoboot disk. @ngtwolf can you please verify?

ngtwolf, 2020-04-19:
So, a few things.  Right now I have an autoboot rom in there (had to break down and do that because originally i didn't know if my trs-io build was bad or just that the frehd autoboot disks didn't work with the trs-io... and after i put the rom in, i was able to confirm it was the autoboot disks that didn't work) so i'm not sure how good my test is, but I did get it working it seems.  A few things though:

1) line 30 should be 'THEN STOP".  

2) The first set of code seems to work, if I do the system and the /20480 command... However the additional code (50/60/70) either doesn't do anything or gets an illegal function call at 70.

3) It works on LDOS 5.3.x, but doesn't work on lsdos 6.x (similar to rsclient).  I think that's a moot point though because if It can be made to work on a 5.3 disk, that would work for boot purposes anyway and would support both model 3/4.  

4)  I also noticed that on my Model 4 GA, that line 30 works with either 3 or 4.. so in theory a LDOS 5.x boot disk could be made to support either Model 3 or 4 with the same program.  Of course, just need to figure out how to do this with some autoboot disk rather than the basic/system command.

Anyway, It would be a pain to confirm this works with the original rom since i'd have to open it back up, but I guess I could do that if we could have a working bootable disk to try that would boot to the frehd.  

Thanks!

apuder, 2020-04-19:
Couple of quick questions: did you use a M3 or M4 for your tests? Also, where did you get the autoboot floppy from? Do you happen to have the source of whatever is on that autoboot disk?

ngtwolf, 2020-04-19:
Mentioned in my comment above, but it's a Model 4 Gate Array (which is why i originally couldn't do a boot rom since it's a pain to do on the gate array since it needs a special board that I had to have made which took a while and was hoping i could get by without doing that at least to make sure everything worked).  

The autoboot disks I thought I got on ian mavrics site at one point but I can't ever find anything there, so heres the creators site (FreHD Boot Floppies file):

https://desertsagesolutions.com/sw/

apuder, 2021-12-25:
Fixed in commit 37639e0c01467c2f105af56a36e599fbf7b4413e.

closed 2021-12-25 by apuder