apuder/TRS-IO #25: Feature Request #2 - Floppy Drive emulation

Howlinmoon, 2020-11-27:
Looking at the FreHD, it appears to only emulate hard disk images, but not floppy drives.
There are a wealth of floppy disk images (protected or non-standard booting) in the DMK format that are not easily transferred to a hard disk image.

Since it appears that all of the required port and address lines are available to the TRS-IO, this feature request is to add floppy drive emulation.
Surely the ESP32 has enough computing power to handle reading and writing to emulated floppy disk images.
This would be similar in functionality to the popular Gotek type floppy disk emulators - but built nicely into one package.

Mounting the floppy disk images could be handled in the same fashion as the hard disk images for the FreHD - out of band via either the on screen configuration - or even via the web interface.

apuder, 2020-11-27:
That is a nice idea but most likely not possible. I/O cards such as TRS-IO or FreHD assert the WAIT* pin to give themselves enough time to finish a request. This works well for in/out operations on the I/O port; but it does NOT work for memory mapped I/O. This is a peculiarity of the Z80 CPU: a WAIT* signal can be generated for memory read requests, but NOT write requests. If you look at the timing diagrams for the  Z80 memory access you can see the reason why. Bottom line: An I/O card cannot intercept memory mapped write access. That is what would be needed to emulate floppy disks, so I don't see a way to make that happen. Quick note on the auto-boot feature of TRS-IO for the M1: you may wonder why that works (since the auto boot is also floppy based). I checked the ROM code that the M1 uses to load the boot loader and just by intercepting read requests, I can "fly blind" and return that the ROM expects. But if you begin to issue other FDC commands, my autoboot code will no longer work.

closed 2020-11-27 by apuder

Howlinmoon, 2021-02-19:
As per TrashTalk Live #11 - I see you have changed your mind...
I knew it had to be possible - since the original expansion interface was working and it has only the same edge connector signals available to it as TRS-IO does.
I look forward to the board revision that will enable intercepting the memory writes.