apuder/TRS-IO #26: Feature Request #3 - Printer Emulation

Howlinmoon, 2020-11-27:
To finish out a complete package, if my other feature requests were implemented, the only feature lacking from the complete expansion solution is a printer port emulation.

Arguably, the majority of printing performed back in the day was simple text printouts from within programs and LLISTings of the current BASIC program.

This would just require a buffer to capture the text sent to the emulated printer port - which could then be retrieved and downloaded via the web interface as a simple text file.

That would suffice for probably 90% of the use cases, to handle the other 10% (just random percentages I'm making up) going above and beyond would be to emulate something like the Epson MX80's graphics codes and other special characters - and instead of creating a simple text dump - emulate how the printer would output the data by creating a downloadable PDF file from the web interface.

I personally, would be happy with a plain text capture / dump mechanism - but if you are feeling extra motivated... :-)

apuder, 2020-11-27:
How does the M1/MIII access the printer? Via the I/O port or memory mapped access? If the latter, it won't be possible (see my comment for the FDC support)

apuder, 2021-12-25:
The latest version does support printer emulation. Visit trs-io.local/printer then use LPRINT or LLIST. The web interface is barebones. PRs welcome. :)

closed 2021-12-25 by apuder