apuder/TRS-IO #34: SD card not working

timgale101, 2022-07-26:
The SD card status on the TRS-IO Configuration page shows up red, when I hover the mouse over it says "Failed to initialize SD card".
I've tried a couple of SD cards, both Class 4 formatted as FAT32.

Checked connections of SD card holder to PCB.

WiFi/SMB share works fine.

apuder, 2022-07-26:
Can you post the ESP'S log? (idf.py monitor)

troffon, 2022-07-28:
Unfortunately I'm not familiar with the idf tools.
Attached a log which I got by connecting Putty to ESP32 serial port, hope this will be of use.
Thanks
[ESP32_Putty.log](https://github.com/apuder/TRS-IO/files/9206992/ESP32_Putty.log)

troffon, 2022-07-28:
After seeing that the ESP log failed early on sdmmc_card_init, I decided to re-solder all the sd card contacts.
Now the sd card is functioning correctly.
Closing this issue.

closed 2022-07-28 by troffon