apuder/TRS-IO #36: Is there a Model 4 / 4P variant of this planned in the future?

TyRoden, 2023-01-20:
I see that there is a fork of this project for the model 4 / 4P, but it seems to be old at this point an perhaps unmaintained. Since this main project seems to be moving along well - are there plans to make this available to us with Model 4 / 4Ps? There is a bit of envy from those of us that do not have a Model 1/III

Thank you.

apuder, 2023-01-23:
Which fork are you referring to? I am not sure about the 4P, but I believe that TRS-IO for the MIII also works for the M4. I don't have a M4 so I cannot test.

TyRoden, 2023-01-24:
This is the one I was looking at - which seems to support the 4 (and I
would imagine the 4P), though it is seemingly quite a bit behind your
commits in its fork path.

https://github.com/pski/trsnic

I'm just hunting a good solution for my 2 model 4Ps. Much appreciated.

apuder, 2023-02-26:
To my knowledge TRS-IO for the MIII also works on the M4. Pete's fork added a networking API that I since merged into this repo. I am not aware that @pski added anything specific to the M4.

closed 2023-02-26 by apuder

TyRoden, 2023-02-27:
Much appreciated