apuder/TRS-IO #37: SDK version - Eclipse version - M1 non-xray w/sd Q

dhlav, 2023-02-11:
Hello, just wondering what SDK (expressif) version is currently being used and the same for eclipse for this project. I couldn't find any reference (I may have missed it). 

I also just built my board using the m1 non-xray with sd card. Is that going to be still functional with the current master branch? I do see the pictures at root of repo have changed.

If I get this right, I can help document the process to regenerate/build.

Thanks! Dave.

dhlav, 2023-02-24:
I was able to rollback to a commit that works now. I used 4.4.2 for esp. Eclipse I didn't use. I'm assuming fpga going forward from now on.

apuder, 2023-02-26:
Sorry for the slow response. Looks like you figured out the issue yourself. The current master no longer works on the non-FPGA versions. Since you manage to get it to work, it would be nice if you could send a PR with the version number as the branch name. E.g., trs-io-m1-1-0 or something like that. Then you can cherry pick fixes from master. But we don't plan to put any more effort into the non-FPGA version, FYI. Please close this comment if you have no further question.