apuder/TRS-IO #39: Documentation provided is incomplete

FreddyKElmSt, 2024-01-07:
I want to build one of these from scratch.  I need details on programming; Tang and ESP32.

If there's already instructions then they are not logically named (ie 'Tang programming') and we can't find them.

Nephrite-FM, 2024-01-07:
Yes, this would be helpful for me as well, if you could point us in the right direction for getting your VHDL code into the Tang NANO 9K and the right code into the ESP32 Module that would be helpful...

apuder, 2024-01-07:
No doubt more documentation would be useful. The ESP32 source is in src/esp. It needs to be first configured via idf.py menuconfig followed by idf.py build. This is the standard way to build an ESP project. The FPGA src is in src/fpga. Always use the latest version (we keep old versions in separate directories). You have to download the Gowin SDK to compile the Verilog and flash to the 9K. This is also documented in the Gowin SDK.

closed 2024-01-07 by apuder