apuder/TRS-IO #41: ESP idf.py will not compile with ESP-IDF 5.1

Nephrite-FM, 2024-01-07:
ESP idf.py will not compile with ESP-IDF 5.1.2, and I have noticed that previous issues you closed out, for earlier versions of the ESP32 code you suggested you need to make and compile the ESP binaries with an older version of the IDF, so for the latest version you have posted here, which version do you need?  3.2.2 again? $.2.5 also fails to get past 'idf.py menuconfig':

CMake Error at C:/Espressif4.2/frameworks/esp-idf-v4.2.5/tools/cmake/build.cmake:189 (message):
  Failed to resolve component 'libsmb2'.

apuder, 2024-01-07:
Please use ESP-IDF 4.4.4.

closed 2024-01-07 by apuder

Nephrite-FM, 2024-01-08:
That did the trick on Debian 11, but not completely.  ESP-IDF Compiles in Debian, however it still fails with this partition problem with the TRS-IO git clone:

[3/1232] cd /home/vboxuser/esp/TRS-IO/src/esp/bu...xuser/esp/esp-idf/tools/cmake/scripts/fail.cmake

FAILED: esp-idf/trs-io/CMakeFiles/spiffs_elFinder_bin 

cd /home/vboxuser/esp/TRS-IO/src/esp/build/esp-idf/trs-io && /usr/bin/cmake -E echo "Failed to create SPIFFS image for partition 'elFinder'. " "Check project configuration if using the correct partition table file." && /usr/bin/cmake -E remove /home/vboxuser/esp/TRS-IO/src/esp/build/esp-idf/trs-io/GHGVx.cmake && /usr/bin/cmake -E env "FAIL_MESSAGE=Failing the build (see errors on lines above)" /usr/bin/cmake -P /home/vboxuser/esp/esp-idf/tools/cmake/scripts/fail.cmake

Failed to create SPIFFS image for partition 'elFinder'.  Check project configuration if using the correct partition table file.

CMake Error at /home/vboxuser/esp/esp-idf/tools/cmake/scripts/fail.cmake:3 (message):

  Failing the build (see errors on lines above)

[5/1232] cd /home/vboxuser/esp/TRS-IO/src/esp/bu...xuser/esp/esp-idf/tools/cmake/scripts/fail.cmake

FAILED: esp-idf/trs-io/CMakeFiles/spiffs_roms_bin 

cd /home/vboxuser/esp/TRS-IO/src/esp/build/esp-idf/trs-io && /usr/bin/cmake -E echo "Failed to create SPIFFS image for partition 'roms'. " "Check project configuration if using the correct partition table file." && /usr/bin/cmake -E remove /home/vboxuser/esp/TRS-IO/src/esp/build/esp-idf/trs-io/9Qhpu.cmake && /usr/bin/cmake -E env "FAIL_MESSAGE=Failing the build (see errors on lines above)" /usr/bin/cmake -P /home/vboxuser/esp/esp-idf/tools/cmake/scripts/fail.cmake

Failed to create SPIFFS image for partition 'roms'.  Check project configuration if using the correct partition table file.

CMake Error at /home/vboxuser/esp/esp-idf/tools/cmake/scripts/fail.cmake:3 (message):

  Failing the build (see errors on lines above)

[7/1232] Generating ../../partition_table/partition-table.bin

Partition table binary generated. Contents:

*******************************************************************************
# ESP-IDF Partition Table
# Name, Type, SubType, Offset, Size, Flags
nvs,data,nvs,0x9000,24K,
phy_init,data,phy,0xf000,4K,
factory,app,factory,0x10000,1536K,
html,data,spiffs,0x190000,200K,
*******************************************************************************

[8/1232] Building C object esp-idf/mbedtls/mbedtls/library/CMakeFiles/mbedcrypto.dir/dhm.c.obj

ninja: build stopped: subcommand failed.

ninja failed with exit code 1

make[1]: *** [Makefile:9: all] Error 2

make[1]: Leaving directory '/home/vboxuser/esp/TRS-IO/src'

make: *** [Makefile:8: all] Error 1

Nephrite-FM, 2024-01-08:
Ok, it compiled all the way for me in ESP-IDF 4.4.6 for Windows.  Perhaps Debian didn't clean out ESP-IDF 4.2 fully, So I'll try recreating the Debian steps with 4.4.4 in a fresh Debian 11 VM.

I got to the point where it wants me to flash the ESP32 chip, so if I have to change settings around, I can go into the idf.py menuconfig and fix whatever might be broken.

apuder, 2024-01-08:
Make sure to do idf.py menuconfig and select the Model III.

Nephrite-FM, 2024-01-09:
> Make sure to do idf.py menuconfig and select the Model III.

Will do, thanks

Nephrite-FM, 2024-02-25:
This did nothing.  Although 4.4.6 appeared to flash the ESP32 Module, when the ESP32 is in the board and connected to the Model III, we do not get a white LED, nor do we get a SSID entitled "TRS-IO", so clearly the flashing didn't work or what was flashed to the ESP32 was not properly compiled.

Where are the instructions that walk users through compiling this, and pointing us to the right resources to compile this?

Nephrite-FM, 2024-02-25:
You should also be aware that there was no ESP-IDF 4.4.4 Release, nor can I get it to compile again.  The process keeps returning different errors every single time, terminating itself at random points through the compile process.

Try 1 - build fails at 3/1235 and 5/1235
Try 2 - build fails at 625/1222 and 627/1222
Try 3 - build fails at 2/595 and 4/595
and so on...

apuder, 2024-02-26:
Please try the last stable release of the v4.4 (which is 4.4.6)