apuder/TRS-IO #43: ESP32 Will not respond to flashed ELF, provide SSID or even turn the LED White

Nephrite-FM, 2024-02-25:
We have recently built your board, flashed the ESP32 and the Tang Nano.  Although 4.4.6 appeared to flash the the ESP32 Module over USB successfully, when the ESP32 is in the board and connected to the Model III, we do not get a white LED, nor do we get a SSID entitled "TRS-IO", so clearly the flashing didn't work or what was flashed to the ESP32 was not properly compiled.

You make a suggestion in an earlier issue that items in the src folder need to be built with MAKE.  What precisely needs to be compiled here?

Where are the instructions that walk users through compiling this, and pointing us to the right resources to compile this?

FreddyKElmSt, 2024-02-25:
I'm also having this same issue

apuder, 2024-02-26:
Please check the ESP logs (idf.py monitor). Did you also flash the 9K? The ESP tries to find the FPGA and will only continue with the boot process once it found the 9K. That would be my guess, but hopefully the logs will give a hint on what is going on.

Nephrite-FM, 2024-03-02:
it says SPI: FPGA Not Found, repeatedly.  I'll try reflashing the Tang Nano...

Is there some reason GOWIN says it is flashed when it isn't actually flashed?  How can we test the Nano is actually flashed?

Nephrite-FM, 2024-03-02:
Digging on the Web, the latest GOWIN has Problems with the TANG boards, I'll try flashing it with Education version 1.9.8.03 others recommend...

Nephrite-FM, 2024-03-02:
Ok, OpenFPGALoader, once I got it loaded with its missing dependencies in MSYS2, flashed the Tang NANO, and the ESP32 says it has found it...  Now, however it just flashes BLUE-GREEN-WHITE-OFF-WHITE at us, while still not broadcasting an SSID.

it gets this far:

I (1061) SPI: Found FPGA
I (1065) gpio: GPIO[4]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0
I (1074) gpio: GPIO[5]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0
I (1083) gpio: GPIO[32]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0

and then stops.

apuder, 2024-03-02:
Just to be sure, you did run `idf.py menuconfig` and selected the right TRS-IO model, right?

Nephrite-FM, 2024-03-02:
Correction - I Recompiled the ESP-32 with the Windows toolchain disabling the LED Test, and now it is flashing Cyan repeatedly, and broadcasting an SSID.  Apparently the RED LED line isn't connected right, I'll look into that later...

The only trouble now is that it is failing to read an SD Card (How do we need to format it?  FAT32 like the FreHD?), and on top of that, the basic snippets meant to retrieve the IP Address fail, even though we get a solid blue light for a few seconds (Still trying to trace down the GPIO Lines).

Yes, we did choose the Model III.

apuder, 2024-03-03:
So, it seems that the Wifi is up and running. Can you connect to the TRS-IO network with your laptop? Can you provide the credentials of your home network? Remember that the ESP can only connect to 2.4 GHz networks. As for the SD card; yes, regular FAT32 should do it. Note that the ESP does not detect insertion or ejection of the SD card. You have to reboot the ESP if you do that.

Nephrite-FM, 2024-03-03:
We have tried rebooting and even hard-powering and it still fails to recognize the SD Card.  We were able to connect it to the home WiFi, and what happens when it connects, GPIO32 lights up - Your board has GPIO32 of the ESP32 running to the Blue LED, and even your KICAD has that pin marked LED-Blue.

Another glitch - GPIO5 is not connected to the RED LED line as in your circuit or code, the way the pinout is, Pin 32 of the ESP32 is actually GPIO21.  I am trying to go through menuconfig to correct GPIO5 to be GPIO21.  GPIO4, as wired, does run to the green circuit (PIN 26).  We were sure to get the HiLetGo ESP32 module, but for some reason the pinout is wrong.  It is very likely that other GPIO Lines are also incorrect...

Nephrite-FM, 2024-03-03:
OK, I fixed GPIO5 to be GPIO21, and the LED Test is now faithfully flashing RED-BLUE-GREEN-WHITE-OFF-WHITE.

SO, Now, I will need to figure out from the circuit what GPIO Lines are supposed to be connected to the TANG NANO 9K - it can't be all wrong as the ESP32 is detecting the FPGA...

Nephrite-FM, 2024-03-03:
The trouble is, if the pinout is indeed incorrect, precisely which HiletGo module should we get?  Because I don't see in the configurator how we may set the I/O lines to the FPGA...

apuder, 2024-03-03:
The LED is connected to different pins for the M1 and M3. You are correct that you have to manually configure the correct GPIO pins in menuconfig. Actually, I removed that option in the latest version and it will automatically pick the right pins. Perhaps do a git pull. As for the SD card: it seems that SPI is working, so the one thing that might not be OK is the CS line for the SD card (the SD card has the same SCK, MISO and MOSI)

Nephrite-FM, 2024-03-03:
while trying to build it with the OTA enabled, even with the new git, we get this:
C:/ASS/trs-io/src/esp/main/ota.cpp:3:10: fatal error: storage.h: No such file or directory
 #include "storage.h"
          ^~~~~~~~~~~
compilation terminated.

so I leave it disabled - and we get a flag indicating FreHD, however the RetroStore and FreHD Functionality still do not work.

The IP Address Basic code does work, although it pulls the wrong IP address...  So I believe the next step for us is to see if we've burnt out one of the tranceiver/translator lines, or there might be a short somewhere - although I did tone-test it out, there is still something amiss - It reports its IP as 082.068.0.060 but the Modem says it is 192.168.1.171 - so, right off the bat we can tell that bit 1 is not getting through...

apuder, 2024-03-03:
Please do not select either OTA or RetroStore options. You wrote that "It reports its IP as 082.068.0.060". The ESP logs the IP address in decimal, without leading 0s. Where did you see this? Please check the ESP logs. In case you use Discord, we can move the discussion here: https://discord.gg/pJPHn8K5

Nephrite-FM, 2024-03-03:
We see 082.068.0.060 when running this on the TRS-80:

10 OUT 236,16
20 OUT31,0:OUT31,4
30 C=INP(31):IF C=0 THEN END
40 PRINT CHR$(C);
50 GOTO 30

Nephrite-FM, 2024-04-11:
Ok, turns out it was faulty wiring.  One of our Translator-Transceivers has 2 pins shorted, which would screw up te Data and Address buses.

airgunning off the chips and reseating them seems to have fixed a number of issues - except for some reason, unless an SD Card is inserted, Wi-Fi is present and a valid SMB share is defined, it will not initialize and Identify itself to the TRS-80. 

We can use the latest Version of Gowin to Compile the nano9k VHDL, but I used MSYS2 + OpenFPGALoader to get it on the Tang nano.  The ESP-IDE 4.4.6 Toolchain for windows did work, and there were a few minor code tweaks to get it going enough to where FreHD and the RetroStore started Working by having it ignore the status of SMB - so looking forward to the next Git Release.

closed 2024-04-11 by Nephrite-FM