apuder/TRS-IO #50: html size in partitions.csv is too small

magore, 2024-12-07:
The html section of partitions.csv has to be about 260K as of today Dec 6 2024 otherwise the project will fail to build running out of space when spiffsgen.py  is invoked during build
This works:
# Name,   Type, SubType, Offset,  Size, Flags
# Note: if you have increased the bootloader size, make sure to update the offsets to avoid overlap
nvs,      data, nvs,     ,        0x6000,
phy_init, data, phy,     ,        0x1000,
factory,  app,  factory, ,        0x180000,
html,     data, spiffs,  ,        260K,

lkesteloot, 2024-12-07:
Weird, it works for me. When did you last pull? Be sure that you have commit 6855eb5a, which reduces the number of things in the html partition.

apuder, 2024-12-07:
I think @magore wants to compile for TRS-IO-M1; not for TRS-IO++. I have not been good at making sure the current code base still works for TRS-IO-M[1/3]. TRS-IO++ has its own partition table (partitions-pp.csv) while TRS-IO-M[1/3] uses partitions.csv. The latter should be updated to adjust the size of the html partition to whatever is currently in partitions-pp.csv

magore, 2024-12-07:
I did the pull this evening when I noticed the portion overflow. I changed 0x32000 to 0x3d000 to resolve the overflow. You can run the command by hand to check it. Something like this 

/share/embedded/CLASSIC_COMPUTERS/trs80/TRS-IO/.espressif/python_env/idf4.4_py3.10_env/bin/python \
        /share/embedded/CLASSIC_COMPUTERS/trs80/TRS-IO/esp-idf/components/spiffs/spiffsgen.py \
        0x32000 \
        /share/embedded/CLASSIC_COMPUTERS/trs80/TRS-IO/TRS-IO/src/esp/html \
        /share/embedded/CLASSIC_COMPUTERS/trs80/TRS-IO/TRS-IO/src/esp/build/html.bin \
        --page-size=256 --obj-name-len=50 --meta-len=4 --use-magic --use-magic-len

commit 5452cfa9 (2024-12-16): Compress files in HTML partition to fit in SPIFFS.

This commits makes several changes:

1. Splits sdkconfig.trs-io-pp into three files, one for
   TRS-IO M1, one for TRS-IO M3, and one for TRS-IO++.

2. Pulls their common elements into a generic TRS-IO sdkconfig file.

3. Renames the non-PP partition file to m1-m3 for clarification.

4. Shrinks the PP "html" partition to 200 kB (from 2 MB) to match the
   non-PP one, so that we can catch errors while developing for ++, and
   to make flashing faster.

5. Updates the Makefile to use the right sdkconfig files depending
   on the BOARD variable  [... cut]
closed 2024-12-16 by lkesteloot